epics-base/pvDataCPP #27: Truncation in BitSet::or_and()

mdavidsaver, 2016-02-17:
For "this |= set1 & set2" the result size should be "max(this, min(set1, set2))" while previously it was "min(set1, set2)" resulting in truncation if the LHS is longer than the RHS.

Practically this can lead to truncation of the overflow bit set, clearing bits which were previously set.

As the fix is straightforward I've already pushed d4292d81f2391a5bb5e5747f0f6ceacec028c1e2 along with an added test case.

The bug was present in the initial implementation.

mdavidsaver, 2016-02-17:
@dhickin How do you want to track fixes which should go into a bug fix release?

msekoranja, 2016-02-17:
+1 on a fix.

dhickin, 2016-02-18:
Fixed by d4292d81f2391a5bb5e5747f0f6ceacec028c1e2

closed 2016-02-18 by dhickin