epics-base/pvDataCPP #43: testTypeCast.tap: 8-bit integers printed as non-printing characters

dhickin, 2016-09-07:
The testTypeCast.tap file from APS Jenkins build server contains a number of instances of 8-bit integers being printed as characters.

In particular a (u)int8_t with 1 is printed as ascii 0x1 = start of heading (SOH).

Note: The latest Jenkins build reports TAP parse errors in this file and the build is being subsequently marked as unstable. This may be related to the non-printing characters. There are 13 SOHs in the tap file and 13 exceptions in the parsing.

dhickin, 2016-09-07:
I've pushed a temporary fix by escaping the raw output. This doesn't affect the test results, just the strings outputted.

With this fix the APS Jenkins build is now stable. (This was the mac build.)

Michael, since this is your test/code could you look at it?

mdavidsaver, 2016-09-07:
Should be fixed with da8ba56dd1cd87c532b5fa625644151b655005fe

anjohnson, 2016-09-10:
Weird, the number of tests passing on APS Jenkins went down from 10120 to 2024. Previousy builds had "140 TAP test set(s), and 0 parse error(s)"; the build after Michael's fix has "28 TAP test set(s), and 0 parse error(s)." It appears the previous the TAP plugin was counting each .tap file 5 times though, so the current number is right. Perl counts 28 test programs and 2025 tests instead of 2024 for me, but that's close enough for now.

closed 2016-09-10 by anjohnson

anjohnson, 2016-09-14:
Merged Michael's fix onto the release/6.0 branch.