epics-base/pvDataCPP #52: Field sub-class comparison is incomplete

mdavidsaver, 2018-02-07:
When the specializations BoundedScalarArray, FixedScalarArray, and BoundedString were added, the comparisons (operator==) were not updated.  So only the element/scalar type is to decide on type equivalence.

I'm not sure what the technical effects of this omission are, but given that no code seems to use these it's minor.

mdavidsaver, 2018-10-16:
@ralphlange reports that a crash, probably due to this issue, has been observed at ITER when two FixedScalarArray arrays are used with differing sizes.  This is with 7.0.1.1

mdavidsaver, 2018-10-17:
Given that this feature has had ~zero use in the past 4 years, I'm inclined to deprecate it.  That is, not just the class definitions, but eventually the protocol feature.  Of course this isn't a step to be taken lightly, so it would need to be done in steps.

1. Mark BoundedScalarArray, FixedScalarArray, and BoundedString as deprecated
1. Hide the implementations.  This would prevent a node from creating structures with these types, but allow it to handle definitions received from a peer.
1. With a minor version bump, begin disallowing for newer versions
1. Drop support along with minor version 1 (probably as I retire)

mdavidsaver, 2018-10-17:
Marked "help wanted" as I'm not inclined to spend time fixing this feature.

dirk-zimoch, 2021-06-17:
In my opinion it would make sense to disable the EPICS_DEPRECATED macro for EPICS internal calls. At least for building the tests.

ldoolitt, 2024-02-22:
Hello, @mdavidsaver !  Fancy meeting you here.  :-)
Is it expected to see six of these deprecation warnings when building epics-base-7.0.8?
In 2024, over five years since you deprecated them?
At least your deprecation message pointed to a useful place.

dirk-zimoch, 2024-02-22:
Many deprecated functions are still called in tests. Of course it makes sense to test deprecated functions too as long as they still exist. But maybe the deprecation warning can be switched off when compiling tests? (Maybe generally in EPICS base, not only in pvData.)

ralphlange, 2024-02-22:
I would argue against that.

Generally, deprecation warnings are good early warning signs.
We already softened the situation by explicitly not treating deprecation warnings as errors, which would break the build.

Unless we have a way to specifically mask out the deprecation warning in the exact six cases where we know we are calling deprecated functions, I would rather accept a single digit number of false positives than hide an unknown number of real issues.

dirk-zimoch, 2024-02-22:
Maybe then only in test code that is meant to test known deprecated functions do this to silence the warning:
```
#undef EPICS_DEPRECATED
#define EPICS_DEPRECATED
```
Test the deprecated functions last in order not to silence the warning too early.

dirk-zimoch, 2024-02-22:
Stupid me. One needs to do this before `#include`ing the header file with the deprecated functions, but after `#include`ing compilerSpecific.h. Thus it is more of an all-or-nothing choice.

mdavidsaver, 2024-02-23:
> Is it expected to see six of these deprecation warnings when building epics-base-7.0.8?

Yes.  My focus of development in recent years is [PVXS](https://mdavidsaver.github.io/pvxs/).  As PVXS matures, I see the pv*CPP modules as increasingly deprecated in their entirety.

Since no one has complained in the intervening years, I think it is reasonable to proceed with step two (see https://github.com/epics-base/pvDataCPP/issues/52#issuecomment-430482285) and remove these API calls.  As internal APIs, the deprecation warnings could then be removed.

fyi. step 2 (decode but not encode) is currently what PVXS supports wrt. the bounded/fixed array types.