epics-base/pvDataCPP #65: BE array serialization broken

mdavidsaver, 2019-04-08:
The fix for #54 introduced a regression to ByteBuffer effecting serialization to big endian.  The bug appears when an array is serialized by a BE peer, or from a BE peer.  The effect is an out of bounds array access.

mdavidsaver, 2019-04-08:
Thanks to @hjunkes for reporting and identifying the cause.

https://epics.anl.gov/tech-talk/2019/msg00571.php

mdavidsaver, 2019-04-08:
Added test coverage with 5525119778bcefdbe53103a1c07984c0d7fae112 and fixed with 8f0111e482aa139f4311378f2b0bc831c46420b2

hjunkes, 2019-04-09:
I have tested the fix on RTEMS - 5 PowerPc
It worked. The test also worked.

mdavidsaver, 2019-04-09:
Re-opening until this fix is included in a release.

mdavidsaver, 2019-08-07:
Released in module version 8.0.0, included in EPICS 7.0.3

closed 2019-08-07 by mdavidsaver