epics-base/pvDataCPP #66: Mismatched docs/implementation for getField

brunoseivam, 2019-05-13:
The documentation for `Structure::getField(size_t)` and `Union::getField(size_t)` claim that they return a null pointer if the index is not valid. The implementation throws a `out_of_range` exception instead (from [`vector::at`](http://www.cplusplus.com/reference/vector/vector/at/)). 

Also, `Union::getField(size_t)` returns `FieldConstPtr` but `Union::getField(size_t)` returns `const FieldConstPtr &`. I think both should return `FieldConstPtr` for consistency.

`Structure`:

https://github.com/epics-base/pvDataCPP/blob/cd2436342d71b173eb4377e61d396e348c15546a/src/pv/pvIntrospect.h#L734-L740

`Union`:

https://github.com/epics-base/pvDataCPP/blob/cd2436342d71b173eb4377e61d396e348c15546a/src/pv/pvIntrospect.h#L857-L863

brunoseivam, 2019-05-13:
Potentially fixed via #67

mdavidsaver, 2019-05-13:
Looking back, it seems that `Structure::getField(size_t)` has never returned NULL.

It's original form in 2012 (87bff33c302a16ff4b4716fd6d5538a9e10af7c7) was

```c++
FieldConstPtr getField(std::size_t index) {return fields[index];}
```

And it is now:

```c++
const FieldConstPtr& getField(std::size_t index) const {return fields.at(index);}
```

So originally, out of bounds access was simply undefined (probably SIGSEGV).  Now it throws an exception.

PR #67 (2019-05-20): Add {Structure,Union}::getFieldT, fix {Structure,Union}::getField
Fixes #66
closed 2019-05-20 by mdavidsaver