epics-base/pvDataCPP #68: Problem building pvDataCPP with c++11 on macos

ddamiani, 2019-06-30:
I noticed a problem build pvData on macos when building with -std=c++11. It seems to be an issue with PVValueArray templates. It builds fine for me with clang and c++11 under linux.

Output from the build failure of testApp/misc/testSerialization.cpp is below. It compiles but the linking step fails:
`c++           -DPVD_INTERNAL    -DUNIX  -Ddarwin     -O3 -g   -Wall    -std=c++11 -stdlib=libc++   -arch x86_64    -fno-common  -I. -I../O.Common -I. -I. -I.. -I../../testApp/misc -I../../testApp/pv -I../../testApp/property -I../../testApp/copy -I/Users/ddamiani/Workarea/epics-base/include/compiler/clang -I/Users/ddamiani/Workarea/epics-base/include/os/Darwin -I/Users/ddamiani/Workarea/epics-base/include -I/Users/ddamiani/Workarea/epics-base/include/compiler/clang -I/Users/ddamiani/Workarea/epics-base/include/os/Darwin -I/Users/ddamiani/Workarea/epics-base/include        -c ../../testApp/misc/testSerialization.cpp
c++ -o testSerialization  -L/Users/ddamiani/Workarea/epics-base/lib/darwin-x86        -arch x86_64            testSerialization.o    -lpvData -lCom   
Undefined symbols for architecture x86_64:
  "epics::pvData::PVValueArray<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >::replace(epics::pvData::shared_vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<signed char>::replace(epics::pvData::shared_vector<signed char const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<char>::replace(epics::pvData::shared_vector<char const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<double>::replace(epics::pvData::shared_vector<double const, void> const&)", referenced from:
      _main in testSerialization.o
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<float>::replace(epics::pvData::shared_vector<float const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<unsigned char>::replace(epics::pvData::shared_vector<unsigned char const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<int>::replace(epics::pvData::shared_vector<int const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<unsigned int>::replace(epics::pvData::shared_vector<unsigned int const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<short>::replace(epics::pvData::shared_vector<short const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<unsigned short>::replace(epics::pvData::shared_vector<unsigned short const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<long long>::replace(epics::pvData::shared_vector<long long const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
  "epics::pvData::PVValueArray<unsigned long long>::replace(epics::pvData::shared_vector<unsigned long long const, void> const&)", referenced from:
      (anonymous namespace)::testArray() in testSerialization.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

mdavidsaver, 2019-06-30:
Can you try adding some more explicit template instantiations?

https://github.com/epics-base/pvDataCPP/blob/c2bc77a6498dfe6f1cad1b0a28680e81764779cc/src/factory/PVDataCreateFactory.cpp#L747-L751

eg. add:

```c++
template class PVValueArray<boolean>;
template class PVValueArray<int8>;
template class PVValueArray<uint8>;
```

And see if any of the errors go away.

ddamiani, 2019-06-30:
That seems to have done the trick! I can submit a pull request with the changes if you like?

mdavidsaver, 2019-07-01:
Please do!

closed 2019-07-08 by ddamiani