epics-base/pvDataCPP #72: WIN32: debug assertion "failures"

mdavidsaver, 2020-06-02:
Seen during an appveyor run (`Environment: CMP=vs2013; Configuration: dynamic-debug; Platform: x64`).  The test didn't register any failures (all `ok`).

https://ci.appveyor.com/project/mdavidsaver/epics-base/builds/33248176/job/i165b2y5005ufg8i#L12543

```
perl -CSD testPVScalarArray.t -tap > testPVScalarArray.tap
Dumping a stack trace of thread 'win58c':
[  00007FFFCDB2D0D5]: c:\projects\epics-base\modules\libcom\src\osi\os\win32\osdbacktrace.cpp[line 22](epicsBackTrace+0x45)
[  00007FFFCDB2CD76]: c:\projects\epics-base\modules\libcom\src\osi\epicsstacktrace.c[line 79](epicsStackTrace+0x66)
[  00007FFFCDB1C07B]: c:\projects\epics-base\modules\libcom\src\misc\epicsunittest.c[line 80](testReportHook+0x4b)
[  00007FFFBD7C6722](_VCrtDbgReportW+0x6f2)
[  00007FFFBD7BB3B0](_CrtDbgReportW+0xd0)
[  00007FFFBD7BB340](_CrtDbgReportW+0x60)
[  00007FFFCACB65D1](?_Debug_message@std@@YAXPEB_W0I@Z+0x41)
[  00007FFFCDC21647]: c:\program files (x86)\microsoft visual studio 12.0\vc\include\xutility[line 548](std::_Debug_pointer<signed char>+0x47)
[  00007FFFCDC6E5AB]: c:\program files (x86)\microsoft visual studio 12.0\vc\include\xutility[line 2798](std::equal<signed char const * __ptr64,signed char const * __ptr64>+0x5b)
[  00007FFFCDC6D6C9]: c:\projects\epics-base\modules\pvdata\src\factory\compare.cpp[line 159](epics::pvData::`anonymous namespace'::compareArray<signed char>+0xb9)
[  00007FFFCDC6B70B]: c:\projects\epics-base\modules\pvdata\src\factory\compare.cpp[line 204](epics::pvData::`anonymous namespace'::compareField+0x1cb)
[  00007FFFCDC6B152]: c:\projects\epics-base\modules\pvdata\src\factory\compare.cpp[line 339](epics::pvData::operator==+0xc2)
[  00007FF696291CF8]: c:\projects\epics-base\modules\pvdata\testapp\pv\testpvscalararray.cpp[line 93](`anonymous namespace'::testBasic<epics::pvData::PVValueArray<signed char> >+0x118)
[  00007FF696291BA6]: c:\projects\epics-base\modules\pvdata\testapp\pv\testpvscalararray.cpp[line 206](main+0x36)
[  00007FF6962A7F2D]: f:\dd\vctools\crt\crtw32\dllstuff\crtexe.c[line 626](__tmainCRTStartup+0x19d)
[  00007FF6962A805E]: f:\dd\vctools\crt\crtw32\dllstuff\crtexe.c[line 466](mainCRTStartup+0xe)
[  00007FFFD6D313D2](BaseThreadInitThunk+0x22)
[  00007FFFD95854F4](RtlUserThreadStart+0x34)
...
```

mdavidsaver, 2020-06-02:
vs2013 is [`_MSC_VER == 18000`](https://en.wikipedia.org/wiki/Microsoft_Visual_C%2B%2B#Internal_version_numbering).  So I guess the hack needs to be extended.

https://github.com/epics-base/pvDataCPP/blob/0fa927afa7d92f490f92eb4906e27046080309e2/src/misc/pv/sharedVector.h#L544-L555

mdavidsaver, 2020-06-06:
19245ce805162c7b2c35a9b336aa470a500d4a06

anjohnson, 2021-10-01:
This fix was included in the 8.0.3 release in EPICS 7.0.4.1

closed 2021-10-01 by anjohnson