epics-base/pvDataCPP #83: get operation on sofIOC pv don't return timeStamp field

bisegni, 2023-05-05:
Hello EPICS group
I'm working with last epics release and I have notice that making a get operation for a IOC that run with softIOC i don't receive a timestamp field in the structure returned. With the sofIocPVA all work perfectly. I wrong something?

mdavidsaver, 2023-05-06:
see discussion of https://epics.anl.gov/tech-talk/2023/msg00657.php

closed 2023-05-06 by mdavidsaver

bisegni, 2023-05-06:
hi @mdavidsaver, yes he's me. But definitely what could be an effective way to fetch all the information for an PV using CA provider?  could be the best solution, launch two monitor that listen on different fields of the same pv and reassemble the structure on client side?
thank you