epics-base/pvDataCPP #90: sharedVector compile problem in MSVC

zhouyunbin, 2023-07-04:
Hello
     I try to use MSVC compiler v143, v142 or v140 to achieve a client which uses lib pvaClient on windows 10, but the class sharedVector reported errors C7683 and C2860, unable to create reference to void, it looks like MSVC has some big differences with g++, any idea fixes this problem?

because of some special reason on Windows, I could only use MSVC. MinGW is work but lacks of other libraries.

mdavidsaver, 2023-07-04:
Firstly, which epics-base and module versions are you trying to build?

Please provide the error messages as observed.  The file and line numbers are essential information.  If you are unsure, then attach the entire output of `make`.

Also, I do very little windows development.  Please help me to map "v143, v142 or v140" to visual studio releases and/or compiler versions (eg. [`_MSC_VER`](https://learn.microsoft.com/en-us/cpp/preprocessor/predefined-macros)).

fyi. between this repository, and epics-base, we have successful CI builds with VS 2010 through VS 2019.

zhouyunbin, 2023-07-04:
Thank you for your reply so quickly.
The base version I used is 7.0.6.1(including the submodule), the problem I might not describe properly.
compile the entire base is OK(VS2022, VS 2019, and VS 2017), the error is shown when I create a solution then import the dynamic library and include the header files.
the _MSC_VER is 1931( VS2022 17.1)
VS report error file is sharedVector.h, info and line number as list:
error code   |   line number | error info( I translate it into English, maybe not accuracy)
C7683              304                    can not create reference to void 
C7683              305                    can not create reference to void 
C2860              344                    void can not as function parameter
C2860              510                    void can not as function parameter
C2860             616                   void can not as function parameter
C2039           722                    m_vtype not the "****" member
C2039           727                    m_vtype not the "****" member

I have a simple solution for test, if you need I can upload.

> Firstly, which epics-base and module versions are you trying to build?
>
> Please provide the error messages as observed. The file and line numbers are essential information. If you are unsure, then [...]

zhouyunbin, 2023-07-05:
Inspired by your message, I compared cl command mode and IDE mode, use /permissive option successfully compiled, thank you very much.
And I have uploaded the library to the nuget.org, it's convenient for everyone to use. https://www.nuget.org/packages/EPICS

closed 2023-07-05 by zhouyunbin