epics-base/pvDataCPP #92: LP: #2029482 — pvget json error

anjohnson, 2023-08-05:
In [LP: #2029482](https://bugs.launchpad.net/epics-base/+bug/2029482) Matt Clark reports that `pvget -M json` doesn't print fields containing NaN values and generates an error "Error in monitor handler : yajl_gen_invalid_number" instead.

The simplest fix would be for the json formatter to just output json5 (available when pvDataCPP is built against 7.0.6.1 or later), in which case the patch I posted to the LP bug above implements that.

A possible alternative would be to add a separate `-M json5` formatter option which enables the json5 option; properly supporting NaNs and Infinity in regular json would require converting them as strings.

mdavidsaver, 2023-08-06:
> ... the patch I posted to the LP bug ...

It isn't much use there.  Maybe a PR?

anjohnson, 2023-08-07:
It may be useful to the bug's reporter, I didn't have time to do the PR yesterday.

closed 2023-08-09 by anjohnson